bigbrodude6119/flipper-zero-evil-portal #5: Audrino Fatal Error

Jayvolution, 2023-07-03:
Hi, i cant upload the Evilportal file on my dev board, BC audtino IDE says: 

Sketch uses 722802 bytes (55%) of program storage space. Maximum is 1310720 bytes.
Global variables use 44804 bytes (13%) of dynamic memory, leaving 282876 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM12

A fatal error occurred: Could not open COM12, the port doesn't exist
Failed uploading: uploading error: exit status 2

any fix for that ?

bigbrodude6119, 2023-07-03:
Hello, it looks like your computer isn't able to communicate with your board. 

Can you try unplugging your board, holding down the boot button, plugging in your board, and then releasing the boot button? 

Then you can try the upload process again.

Jayvolution, 2023-07-03:
it does communicate, i can earase Maurader and installed, and audrino IDE is showing that its connected

Jayvolution, 2023-07-03:
imma try to restart and removing some us b devices

bigbrodude6119, 2023-07-03:
Interesting. Ok let me know if that ends up working.

Jayvolution, 2023-07-03:
does this seem right ? 
Leaving...
WARNING: ESP32-S2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
Failed uploading: uploading error: exit status 1

bigbrodude6119, 2023-07-03:
Yes, that is normal when flashing that board via arduino. Now you can plug the board into your flipper, open the app and it should be good to go.

Jayvolution, 2023-07-03:
okay, i deactivated Windows Defender and it workes, duno why :D but thanks man

bigbrodude6119, 2023-07-03:
Woah that is interesting for sure. I'll have to keep that in mind when having issues. Thanks for working through that 👍

Jayvolution, 2023-07-03:
are there different html files, bc i only the MSN to open ?

bigbrodude6119, 2023-07-03:
Right now there is only the single index.html file. Do you mind clarifying what you mean by MSN to open? I'm not familiar with MSN.

I do plan on supporting multiple portal templates in the future. So ideally you would be able to choose from Google, Facebook, Twitter, etc.

You can always replace the index.html file with your own. Just keep in mind that there are size limitations and you'll need to mimic the form submission that currently exists in the provided index.html.

Jayvolution, 2023-07-03:
its just forwarting to https://www.msn.com/ for me

Jayvolution, 2023-07-03:
and on the phone, nothing happens, just connecting to wifi, did i do smt wrong :D

bigbrodude6119, 2023-07-03:
There should be nothing forwarding you to msn.com. Let me confirm some things with you before going further.

Did you drag the necessary files into the `apps_data` folder? There should be a single `index.html` file, a `ap.config.txt` file, and an empty `logs` folder.

What is the name of the access point that you are connecting to? Unless something was changed it should be "Google Free WiFi".

Jayvolution, 2023-07-03:
i use RM, and i confimed that both files are on my flipper, only the Logs folder is missing

Jayvolution, 2023-07-03:
the local index.HTML file is bringing me to the login site

Jayvolution, 2023-07-03:
imma make a video one sec

bigbrodude6119, 2023-07-03:
Ok, you should create the logs folder then. I don' think it will save logs without it there.

Ok, I think a video might help. Thanks

Jayvolution, 2023-07-03:
https://youtu.be/xa443osELQo

Jayvolution, 2023-07-03:
and on the phone, i can connect, ant thats it, no pop up for login or anything

bigbrodude6119, 2023-07-03:
Thank you for the video! It looks like this is an issue when you have a wired connection on the same machine. You can see others having the issue here: https://community.arubanetworks.com/discussion/guest-portal-redirect-to-msn-issue

For your phone, can you go to your wifi settings and forget the network? Then try connecting to it again within the settings?

Jayvolution, 2023-07-03:
imma try both one sec

Jayvolution, 2023-07-03:
okay works on my pc without wired, but still not on my phone, imma try it od different phones aswell one sec

Jayvolution, 2023-07-03:
Samsung galaxy s23Ultra and S23, does not work

Jayvolution, 2023-07-03:
nothing just "no network"

bigbrodude6119, 2023-07-03:
Very strange. I'll see if any of my family has one of those phones tomorrow. I'll get back to you on here.

Jayvolution, 2023-07-03:
alr :)

bigbrodude6119, 2023-07-04:
Hey so I was able to test it out on some android phones. Unfortunately there isn't much I can do right now in terms of solving the issue. 

It seemed to work when the phone was connected to a different network and then try to connect to the evil network. But when just connecting to the evil network without a previous connection it did not work. 

Obviously this makes the attack less effective so it's a bummer.

Jayvolution, 2023-07-04:
Okay, i could not replicate this with my phone.

bigbrodude6119, 2023-07-04:
Dang. Well thank you for your help diagnosing all of this. I'll do more research but as of right now I'm not sure how to tackle this issue. I'll add it to the issues section of the readme and close this ticket since it started with an unrelated error.

Jayvolution, 2023-07-04:
Alr :)

closed 2023-07-04 by bigbrodude6119