bigbrodude6119/flipper-zero-evil-portal #6: keep getting this issue when flashing, any fix/ workaround?

grantfosterr, 2023-07-03:
WARNING: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
Failed uploading: uploading error: exit status 1

bigbrodude6119, 2023-07-03:
You should be good to plug in the board to the flipper and start the app. This error occurs when on the dev board since it doesn't automatically start after flashing. Please let me know if that works.

grantfosterr, 2023-07-03:
im using a esp32 s2 mini, i put it in the GPIO and the app runs but it jus says "starting portal, if no response press back to return"

bigbrodude6119, 2023-07-03:
Can you provide the pins that you are connecting to the flipper? The app requires 3.3v, ground, Rx, and TX to be connected. 

If you have multiple sets of tx/rx can you try both of them?

grantfosterr, 2023-07-03:
i tried connecting it different ways, that didnt work, could it be a flashing error?

bigbrodude6119, 2023-07-03:
It very well could be. Let's try this:

Connect the board to your PC and open up the Arduino IDE. 

Open up the serial monitor. 

Type 'sethtml=testing' into the text input box and press send. 

If the serial connection is working you should get back "html set". 

Then you can type 'setap=test' and you should see "AP set" and the server should start up. 

If none of this happens we may have an issue with flashing the board. Or you may need to press the reset button on the board.

grantfosterr, 2023-07-04:
i fixed my flashing issue and serial connection works, running the evil portal app and still sats 'starting portal if no response press back'

bigbrodude6119, 2023-07-04:
At that point in the program the board is waiting for the `sethtml` and `setap` command. The app is attempting to send those commands each time you press "Start portal". 

One more thing to confirm, the RX0 pin on the board should go to TX on the flipper and TX0 on the board should go to RX.

bigbrodude6119, 2023-07-05:
Hey, were you able to get this up and running?

moshik132, 2023-07-05:
Thanks man. i have a problem with flashing the Wi-Fi dev board. i see in the arduino this error after the  upload:

Sketch uses 782093 bytes (59%) of program storage space. Maximum is 1310720 bytes.
Global variables use 49788 bytes (15%) of dynamic memory, leaving 277892 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM8

A fatal error occurred: Could not open COM8, the port doesn't exist
Failed uploading: uploading error: exit status 2

can you pls tell me why?
Thanks

bigbrodude6119, 2023-07-05:
Let's try this solution. 

Unplug the board, hold the boot button on the board, plug in the board while still holding the boot button. 

Release the boot button once plugged in. 

Make sure that you have the correct port selected and try again. 

If you are using the marauder firmware on the board already please check out the readme on how to erase that marauder firmware or check out this [youtube video](https://youtu.be/ktIvzB__pTs) by Pen Ace for a tutorial.

bigbrodude6119, 2023-07-07:
@moshik132 I am going to go ahead and close this issue. I released version 0.0.2 with instructions on an easier method of flashing the board.

closed 2023-07-07 by bigbrodude6119