bigbrodude6119/flipper-zero-evil-portal #11: Is there anyway to use this in ESP-32 WROOM ?

canyazbahar, 2023-07-08:
I'm really wondering is there any way to  run this on ESP-32 WROOM.

bigbrodude6119, 2023-07-09:
Hey @canyazbahar, [yes it is definitely possible](https://imgur.com/a/1fTQ9Sl). Right now I am only making pre-compiled builds for the esp32s2 model but the ESP32 Wroom board seems to be a very popular alternative in the community. 

I am going to get some work done on the project tomorrow and I'll see if I can release a pre-compiled version.

If you would like to get it up and running on your board tonight you can follow the option 2 in the readme for flashing the board via the arduino ide.

closed 2023-07-09 by bigbrodude6119

bigbrodude6119, 2023-07-09:
@canyazbahar I have attached a new file to the 0.0.2 release called `esp32_wroom.zip`. These are pre-compiled bins and you can follow the readme guide for the online flasher as if it were the official devboard.

Its not officially supported yet so things like the LED indication doesn't work but everything else should function as intended.

canyazbahar, 2023-07-10:
don't know how much i appreciate it. thx for you hard work bigbrodude69