bigbrodude6119/flipper-zero-evil-portal #14: HTML "login" page not popping up

wrxsnowboarder, 2023-07-09:
When using Xtreme firmware the html login style page is not coming up when connecting to a network.

bigbrodude6119, 2023-07-09:
Hello, are you trying to connect via an android phone? Others have been having issues with some of the Samsung phones.

wrxsnowboarder, 2023-07-09:
Yes samsung phone and toshiba computer

bigbrodude6119, 2023-07-09:
Yeah, I'm not sure on the fix for certain android phones. Unfortunately, I do not have any to test with.

Can you describe what happens with the computer?

wrxsnowboarder, 2023-07-09:
The computer connects to the network but doesn't pull the html page after that. It simply says connected. When clicking on the browser it displays the no internet connection screen.

bigbrodude6119, 2023-07-09:
Hmm that is troubling. Are you running windows? Does the computer have an ethernet cord plugged in while testing this?

wrxsnowboarder, 2023-07-09:
The computer does not have an ethernet cord and is using wifi. It is running on windows 11 os.

bigbrodude6119, 2023-07-09:
Thank you for the details. I'll have to do some testing. Unfortunately I have a busy week ahead of me but I am hoping I can get to the bottom of this issue in the next release.

wrxsnowboarder, 2023-07-09:
Sounds great. I look forward to checking out the update.

hossmcross, 2023-07-20:
> Thank you for the details. I'll have to do some testing. Unfortunately I have a busy week ahead of me but I am hoping I can get to the bottom of this issue in the next release.

Seems Ive got the same issue:

Got it up and running btwn the official dev board and Unleashed firmware. Only tested using Linux so far, Pop!_OS, and no pop-up or prompt to login. My default browser, FireFox, is heavily privacy hardened and its no surprise that it didnt come in; however when trying to hit the Flipper's IP it times out. Also tried hitting the IP with a completely untouched Chrome but to no avail. 

Going to be using a Win10 machine today to see what I can get going, expecting little to no issue there, but all Ive thrown into the flipper is whats outlined in the readme and Im leaning towards a file or two missing from the flipper as Im unaware of anything that may need to be done with the extras in the repo

SkyldaHelvig, 2023-07-30:
Just adding...

Windows 11 worked soon as I opened Chrome.

A cheap android phone did not unless I specifically went to the IP shown on the flipper.

Pixel 7 pro popped up as soon as I connected to the wifi.

Also just tried on Linux Mint and it worked fine as well.

thewettoad, 2023-08-14:
Just adding my experience...
Samsung S23 Ultra running Android 13, One UI ver 5.1. No login page. 
iPhone 14 Pro Max popped up immediately.

kenshiro81, 2023-08-15:
Adding my experience. iPhone 12 used to pop up immediately, today I tried but no working
Win 11 pop up immediatly

gugo2808, 2023-08-19:
Adding my experience. Samsung android 13. Page does not pop out. Just - connected without internet.

nayko79, 2023-08-23:
Yo también probé y no me funciona ni en teléfonos Samsung zfold 4, ni en samsung a 10, ni en pc, ya probé todo y no encontré solución. Estoy usando el unleshead y la placa wifi oficial

nayko79, 2023-08-24:
probe en un smartphone motorola y funciona perfecto, el problema es con samsung y al menos windows 10 que es el que estoy usando

b1n0t, 2023-09-12:
adding my experience: I have the same issue using a iphone 11 or Mac Ventura with Xtreme

anthares101, 2023-09-15:
Hitting this same problem on Android 13 devices, Windows 10 looks like is working just fine.

JohnDoh282, 2024-01-17:
Do you have the correct dev board attributes and libraries uploaded in Arduino? I found it fixed my login interface popup issue but only for every type of device except Android Phones. Apple phones works great, which is most likely going to be the pool anyway.

JohnDoh282, 2024-01-17:
> Hitting this same problem on Android 13 devices, Windows 10 looks like is working just fine with chrome.

Yea, its all android that seems to have a native defensed against this iteration of evil portal.