bigbrodude6119/flipper-zero-evil-portal #26: Junk characters showing in AP name & HTML page

bigbrodude6119, 2023-07-13:
Sometimes when the portal starts the UART connection sends "junk" characters that appear at the end of the AP name and/or in the HTML page. I found that this happens more often when starting the portal quickly after opening the app. 

I'm not sure if its some sort of noise/interference happening during the communication between the flipper & esp32.

leedave, 2023-07-25:
I would recommend switching to FlipperFormat for the AP name configuration. It would change the format of the file but reading out data is pretty easy after that, you always get a key -> value pair out of each line and no random characters. 
Let me know if you need help with it.

aaron-sc, 2023-12-24:
The issue is still occuring!