bigbrodude6119/flipper-zero-evil-portal #27: Support for multiple portals via a file browser & select screen

bigbrodude6119, 2023-07-13:
Allow for the user to select an alternate file to load for the portal & AP config. If the user has multiple HTML files on their flipper this would allow them to change portals without having to use a PC to replace the existing index.html & ap.config.txt.

bigbrodude6119, 2023-07-17:
Portals portion added via PR https://github.com/bigbrodude6119/flipper-zero-evil-portal/pull/41 by
@Nycz-lab  & @NikIsHere. ❤️

Scott1974-1, 2023-07-20:
this would be a great option.

alienatedsec, 2023-08-05:
> Portals portion added via PR #41 

@bigbrodude6119
When do you plan to release 0.0.3 with this feature? dev tree is not building very well

CodyTolene, 2023-08-18:
Potential conflicts here:
https://github.com/bigbrodude6119/flipper-zero-evil-portal/pull/49

Is this better functionality?