bigbrodude6119/flipper-zero-evil-portal #34: Accesspoint not being broadcast

headcase0570, 2023-07-14:
I've used two Android phones, one .acbook, one Windows laptop and a Kobo e-reader. None show the 'Google Free WIFI' network.

When launching the portal, I notice that the Flipper display shows the entire Evil-Portal 'Index.html' file's content. I have Evil Portal as part of the recent Hidden Master release, I did not install it manually. I've been using Marauder on my ESP32 board for a while now, so I know that it's got the right firmware and working correctly. Maybe just a typo somewhere along the HTML code?
I've uploaded a short youtube video demonstrating what I'm seeing on my flipper, link: [evil portal bug?](https://youtube.com/shorts/k9y82b_5gfQ)

Asizon, 2023-07-16:
You need to flash evil portal firmware to the esp32.

Nycz-lab, 2023-07-19:
Look under Releases. There is a full written Tutorial for flashing and the zip archive you're looking for.

Nycz-lab, 2023-07-19:
You won't be able to run Marauder anymore since that requires it's specific firmware on the esp. If you want to run Marauder again you need to flash its firmware again. However there is a tool called "ESP Flasher" which allows you to flash the esp from the flipper itself this way you can do it on the go.

MadPalmTree, 2023-08-13:
I don't think those steps listed above are quite accurate, given this instruction for people running Marauder.... https://github.com/justcallmekoko/ESP32Marauder/wiki/evilportal  Correct me if I'm wrong? I was going strong until I got up to the index.html part, and yet I'm still getting a "could not find /index.html. Use stopscan..." There are a handful of sources on here specifying different things.

jonnyborbs, 2023-12-20:
I have this same issue. I've definitely got the Marauder with Evil Portal firmware loaded (and have tried the Evil Portal only release firmware)

All my configs are in place - the ap.config.txt and the index.html. Both are recognized when starting the portal. The board can scan and capture no problem. But starting evil portal loads the configs, says web server running - but is broadcasting not a single thing from the radio. 

I reloaded Black Magic and confirmed the board isn't broken - it transmits and receives just fine. But for whatever reason, the portal AP never actually stands up.