bigbrodude6119/flipper-zero-evil-portal #39: Portal config inside app + portal presets

AmongusGaming, 2023-07-17:
2 things to add here that would make this evil portal very good.

1: Make it so we can change the name of the WiFi from inside the app.

2: Make it so we can create custom presets so we can create new portals without needing a computer. for example, if i want to switch from a fake Google Login page to a Twitter login page, we should be able to do it with presets. Make it so the name changes based on the phishing page ur using (make us able to change it).

I hope these features can be added in future updates.

jpaulodamico, 2023-07-24:
Essas suas ideias são realmente ótimas!

CodyTolene, 2023-08-18:
Resolved here https://github.com/bigbrodude6119/flipper-zero-evil-portal/pull/49
by @leedave