bigbrodude6119/flipper-zero-evil-portal #65: New Portals Suggestions

Foul, 2023-08-15:
I leave in France and many places have Free Wifi so it would be cool to add/create them, ex:

McDonald's 
SNCF
DisneyLand
...

Feel free to add some more ;)

JMcrafter26, 2023-08-16:
I made a Discord login screen (very realistic)

IMPORTANT: I will release (and pull request) it at the end of the week, because it's not finished yet. Here is a quick Preview: 
[Preview](https://raw.githack.com/JM26NET/flipper-zero-evil-portal/main/portals/discord-assets/Discord%20nano.html)

JMcrafter26, 2023-08-16:
> That's great. Can you help me with the one i am working on? it is not as good as yours :)

I am currently on vacation, but when I am back, I will look into it.

astromax74, 2023-09-15:
Deff need an Instagram portal

JMcrafter26, 2023-09-16:
Ok, I will make it in near future, don’t worry. BUT don’t do silly stuff 😉

JMcrafter26, 2023-09-20:
> Deff need an Instagram portal

HI, it's me again. I finally got to make the Instagram login screen. I tried my best and it looks super realistic.

astromax74, 2023-09-21:
Hi! Did you upload it? It is not un the portals folders, thanks!

JMcrafter26, 2023-09-21:
> Hi! Did you upload it? It is not in the portals folders, thanks!

Oh, I am sorry. I completely forgot to paste the link. Here: https://github.com/JM26NET/flipper-zero-evil-portal/tree/main

gigi74168, 2023-11-14:
I leave in France and many places have Free Wifi so it would be cool to add/create them, ex:

McDonald's
SNCF
DisneyLand
...

Feel free to add some more ;)