bigbrodude6119/flipper-zero-evil-portal #69: the index.html could not find but the file is in the flipper apps data

jas505, 2023-08-22:


leedave, 2023-08-22:
Not very helpful, an issue without description for something that does not happen for other users. 

Maybe this hint will help you, you need the correct subfolder with the files in it. 
`apps_data/evil_portal/`

Kooter3rown, 2024-01-04:
I am having the same issue when I open evil portal are you saying it has to be in another file inside evil portal file on the flipper there is a single index.html file when you first open the evil portal folder which doesn't seem to work for me I get this message saying setting up HTML...
 not found. 
So I make another folder and put the htmls in it 

also can we not download the get files from here with our phone or do we have to pack them ourselves by this I mean anytime I click on the so-called files it just opens into a bunch of text

super new to a lot of this just got my flipper zero for Christmas thanks to my girlfriend and her family if only they knew LOL watched a bunch of videos you can never find the specific information you need thanks a lot everybody

Kooter3rown, 2024-01-04:
Want more thing I forgot to add I know it says this may change and later models but I have the basic Wi-Fi Dev board from flipper zero I don't think it has an SD card slot has that changed do I still need that or is it even possible for me to do this

One more thing I'm running extreme firmware and I have flashed the marauder on my Devboard let me know if I'm missing anything

JohnDoh282, 2024-01-13:
Im in the same boat kooter

akashrpatil, 2024-01-18:
I'm also having the same problem. Could not find /index.html after starting my evil portal. 

- Firmware is updated
- ap.config.txt and index.html files are there in the `apps_data/evil_portal` folder

 let me know if I'm missing or doing anything wrong here