bigbrodude6119/flipper-zero-evil-portal #71: More complex html

ifuchs, 2023-08-23:
Is it within the capability of the evil portal to do more than display the index.html? For example, could another page be shown when the user hits enter?

ifuchs, 2023-08-23:
I tried the following. I changed the index.html to:

`<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Hello</title>
</head>
<body>
    <h1>Hello</h1>
    <button onclick="location.href='world.html'">Go to World</button>
</body>
</html>
`
and added another file (world.html) in that directory:

`<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>World</title>
</head>
<body>
    <h1>World</h1>
</body>
</html>
`
It displays the first file with the button; however, pressing it does not display the second file.

demotedcoder, 2023-08-24:
Or it would be a great idea if you inject js.

ifuchs, 2023-08-24:
I assume you meant, "if you could inject js." or is it possible now? With .js one could probably do in one page what might otherwise require multiple pages.

Omentoak, 2023-08-24:
I tried to create an HTML that looked like a free Wi-Fi with buttons for various platforms (Google, Instagram, and Facebook), but lacking HTML knowledge, I couldn't achieve my goal. I managed to link a button once, but I couldn't replicate the lost scripts. If someone had better HTML knowledge than mine, I'm sure they could succeed.