bigbrodude6119/flipper-zero-evil-portal #77: It would be helpful to be pointed toward what code deals with output parsing (that turns raw output into “u:<email> p:<password>”, to see if I could edit it to accept additional / custom form outputs.

SneakyTaco0110, 2023-10-06:
It would be helpful to be pointed toward what code deals with output parsing (that turns raw output into “u:<email> p:<password>”, to see if I could edit it to accept additional / custom form outputs.

_Originally posted by @wasp604 in https://github.com/bigbrodude6119/flipper-zero-evil-portal/issues/73#issuecomment-1718136048_

SkidFace, 2023-11-04:
The `setupServer()` function within **EvilPortal.ino** is used to parse the data from the form. The username and password are stored to the variables `user_name` and `password` respectively. After this, they are concatenated with "u: " and "p: " in the `loop()`, which sends the data to the flipper by printing to serial.

SneakyTaco0110, 2023-11-04:
> The `setupServer()` function within **EvilPortal.ino** is used to parse the data from the form. The username and password are stored to the variables `user_name` and `password` respectively. After [...]

Would you be willing to help me configure some custom changes to this setup server compensation will be made if so 🤑