bigbrodude6119/flipper-zero-evil-portal #81: It doesn't open the page after a few seconds automatically

Drjacky, 2023-11-05:
> When you connect to the AP a web page will open after a few seconds.

it doesn't open the webpage automatically. but if I put the IP address manually into the Android browser, it displays the fake login page.

kylewa2135, 2023-11-06:
Mine was working perfectly yesterday, tried it this morning and now the portal says “no internet connection” and the webpage does not automatically generate anymore. If I type in the IP I can connect to it however it seems to be an issue now.

Lashaka, 2023-11-06:
@Drjacky  having the same issue

MGodwood, 2023-11-11:
Me too. But not on all phones.

Spessmen14, 2023-12-15:
Having same issue, Samsung model phone.

kmsmith222, 2024-01-03:
I noticed my androids (s8-s10) still have a secondary DNS server (google) when connected to the portal, could this be causing the captive screen not to pop up automatically? 
 I can get to it if I hit "Manage Router" however.  
 If this is the cause, Id suggest a duplicate DNS2 entry since the phone might still be trying to get on the real net.

AngelcoreS, 2024-03-19:
Thank you guys, I went through too many videos and article and I couldn't understand why is not automatically popping the web, I was using my android S21

whitehammerx12, 2024-05-01:
Having same issue, Samsung phones at least.