bigbrodude6119/flipper-zero-evil-portal #88: Bug when changing the portal html

MatteoTaut, 2023-12-25:
When I change to others portals by selecting "Select HTML" on the flipper the wifi html goes bugging.
In the first screen I have the Apple portal and it's all good, when I change it to Amazon they like fuse together, I'm using Xtreme firmware and I've installed the evil portal with flipper katalog.    
[image]
[image]

MatteoTaut, 2023-12-25:
"FIXED" I need every time I want to change the portal to click "reset" on the dev board.

adzsx, 2024-05-12:
Optionally you can reboot the flipper by pressing and holding "back" and "left" before selecting the new html file