bigbrodude6119/flipper-zero-evil-portal #93: ApiTooOld

jacen2110, 2024-01-10:
I followed the "Install the pre-built app on the flipper" and am coming up with the ? icon as well as "ApiTooOld" when trying to run it. 

I have tried a couple different remediation steps but have had no luck, any suggestions?

anil616161, 2024-01-13:
[image]
I have been receiving same error message.

KeksTheAllmighty, 2024-02-03:
any news on this one? am recieving the same message. guess some files in the release section are too old here.

ChrisGehm, 2024-02-04:
Same error here. Any idea on what could solve it?

gauthi3r, 2024-02-20:
@bigbrodude6119 hi, do have plan an update to work with the new API ? thx

chevdor, 2024-02-23:
See also #97