bigbrodude6119/flipper-zero-evil-portal #96: Missingl-mports

Germanope, 2024-02-14:
I'm using the unlshd-071 firmware and I can't use evil-portal, it was working normally and after installing release 0.0.2 it no longer opens the app and always returns the same error (Missingl-mports).

[image]

Gourmeggle, 2024-02-17:
Hey! Did you find an answer to your Problem?

Germanope, 2024-02-17:
Not yet

toy-creator, 2024-02-22:
Hey mates
Any news on how to solve it?

Floppynator, 2024-02-22:
Use the evil portal from marauder. Works fine for me.

Germanope, 2024-02-22:
In the marauder the evil portal never goes up, there is a message indicating the absence of the configuration file, I have already placed this file in several different paths and it always behaves the same way.

[image]

toy-creator, 2024-02-22:
You can try to add file "ap.config.txt" inside dir /ext/apps_data/marauder, remember to add an AP name to your file
But still unable to see evil portal despite my AP name is added 😅

Germanope, 2024-02-22:
I added it to this path, but the damn app thing doesn't recognize it lol
[image]

toy-creator, 2024-02-22:
Maybe your file is not on UTF-8 format or there is any special char on it?

Germanope, 2024-02-22:
I validated it here and it has this encoding, inside it there is only the name of the PA that I left as "Evil Portal", in your configuration file, do you have any other parameters?

toy-creator, 2024-02-23:
> I validated it here and it has this encoding, inside it there is only the name of the AP that I left as "Evil Portal", in your configuration file, do you have any other parameters?

I found this https://github.com/DarkFlippers/unleashed-firmware/issues/582
This probably help you if you try marauder evil portal