bigbrodude6119/flipper-zero-evil-portal #110: MissingImport Error

asafdemir91, 2024-05-02:
Hello , 

I made everything in https://slayer0x.github.io/flipper/ guide . But cant start evil portal . 
I got this error 
[image]
Any Solution ? What am I doing wrong ?

mvanzanten, 2024-05-11:
Same issue here, im on [unlshd-073](https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-073), using latest marauder [v0.13.10](https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.13.10) and trying to use the latest evil portal [evil-portal-0.0.2](https://github.com/bigbrodude6119/flipper-zero-evil-portal/releases/tag/0.0.2), used to work on marauder `v0.13.4`.

eaudaim, 2024-05-17:
Same error here.
Followed all the instrutions 3 times, change firmware to factory and re-install unleashed before tried my 3rd installation try of the evening, but always the same error as the screenshot above, again and again.

[image]

DefinetlyNotAI, 2024-07-13:
Hi, I had this problem, its from the external firmware, Sadly the firmware for some reason MERGES the 2 firmwares and this creates many crashes and import issues, so all you need to do is backup your sd card on your computer, then plug the sd card in ur flipper and go to settings to reformat the sd card, OF COURSE ONLY DO THIS if your sd card storage maximum is weird (like in your computer it says a max file size way different than what is written like 200mb for a 64GB sd card), afterwards you download the firmware as usual (better yet, download the NORMAL official firmware then put the custom firmware ontop of it)