bigbrodude6119/flipper-zero-evil-portal #121: App Too Old

hackerspace09, 2025-01-18:
Whenever I try to use evil portal it gives me an "app to old" error and under that it says "APP:30 <FW:78 this app might not work continue anyways?"

Is there a work around for this? do I need to download an older Unleashed furmware on my flipper if so what firmware works? or dose the creator just need to update it?

SSnoody, 2025-01-28:
i have the same issue

hackerspace09, 2025-01-28:
> i have the same issue

I figured out how to get it to work. I found an OS called Momentum and it comes with Evil portal already installed and then you just have to flash Marauder onto your wifi dev board and it should work after that. lmk if you need more help or if you want me to break it down better. 

the link for Momentum is https://momentum-fw.dev
the link for Marauder is https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher

hackerspace09, 2025-01-28:
> i have the same issue

Also I have a ton of Https files that you can use in evil portal that range from Starbucks login pages to google pages lmk if you want them and ill figure out how to post a zip file

SSnoody, 2025-01-28:
> > i have the same issue
> 
> Also I have a ton of Https files that you can use in evil portal that range from Starbucks login pages to google pages lmk if you want them and ill figure out how to post a zip file

I will see it later, thanks!

lesliesbird, 2025-02-02:
I tried compiling this with the latest firmware and got tons of errors that I don't know how to fix. At least I can use Marauder instead. I wish I knew what changes need to be made.

hackerspace09, 2025-02-02:
> I tried compiling this with the latest firmware and got tons of errors that I don't know how to fix. At least I can use Marauder instead. I wish I knew what changes need to be made.

Can you take some screenshots and show me what the errors are and tell me what what the steps are that you took, like the firmware and witch OS you installed.

hackerspace09, 2025-02-06:
You need to use a Different OS the official operating system runs like it has tuns of training wheels and wont let you use the flipper to its full potential. do you know how to download a different OS? I currently use Momentum. I've found it works with both evil portal and marauder.

This is where I downloaded Momentum from and it works like a dream. 
https://momentum-fw.dev/

lesliesbird, 2025-02-07:
> You need to use a Different OS the official operating system runs like it has tuns of training wheels and wont let you use the flipper to its full potential. do you know how to download a different [...]

The thing is, I compiled it on and it was working with the official firmware before I updated. I'll probably give Momentum a try at some point, but for science's sake I'd like to know what changed.

linkslice, 2025-04-07:
I backported a bunch of fixes, lmk if they work and I'll send a PR over: https://github.com/linkslice/flipper-zero-evil-portal